bigtreetech/BIGTREETECH-TFT35-V3.0 #154: illegal user app

timgray, 2022-08-08:
No matter what I try all I get is a black screen with "illegal user app".  how do I get past this?   I have tried loading the firmware on a 1gb SD card formatted, fat, fat32 and they do nothing.  no card at all gives me this as well.  Would prefer to not return the unit as it did this out of the box, but would like to get it working.

there is a "bootloader fix" but zero instructions on hos to use it.  does it go on a SD card?  use a program?  Use USB? hook it up serially and use a loader program?  What port?  Any details at all on how to make one of these work from.  "Illegal user app" state?

I tried a 4gb card, 1gb card, I even found a 256mb MMC.... all the same.

radek8, 2022-08-09:
What TFT model do you have?
What bin file did you use?
Send a photo of the TFT processor.

timgray, 2022-08-09:
BigTreeTech TFT35-E3 V3.0.1
BIGTREE_TFT35_V3.0_E3.27.x.bin
[image]
[image]

If you really want a blurry blob photo of the processor I can post it, but it can be seen in the above photo.

timgray, 2022-08-09:
I am unsure how these photos will help me plug into the board to re-upload the firmware to try the recovery proceedure.

radek8, 2022-08-09:
These photos helped me figure out that you used the wrong bin file.
You have to use the file
[BIGTREE_GD_TFT35_V3.0_E3.27.x.bin]
(https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0_E3.27.x.bin)

radek8, 2022-08-09:
Bigtreetech has a bit of confusion in product labeling...

timgray, 2022-08-09:
Ok tried that file.   no change at all.   instantly says "Illegal user app!"

timgray, 2022-08-09:
Wait.  It's uploading now.   In frustration I started hitting the reset button rapidly.  it. kept showing the same error over and over and then suddenly  started the loading of the file after the tenth time hitting reset.

So it sounds like rapid fire on the reset until the board stops complaining about an error and actually tries again worked.

radek8, 2022-09-04:
Please close the ticket