bigtreetech/BIGTREETECH-TFT35-V3.0 #156: connection BTTSKR e3 V3 to TFT35 V3.0.1 impossible in Marlin mode

josgar84, 2022-10-14:
Hello 
i'm trying to connect the SKR e3 V2 to TFT35 V3 and i have some issue. 
The connection RS232 works and i can dysplay the touch screen so when i plug the EXP1's board to EXP3's screen it turns black with a loud beep. I have tried EXP1 and EXP2 and it is the same problem. 
I'm looking for and reading a lot of things for a week and i'm going crazy.
My settings in MArlin are " #define CR10_STOCKDISPLAY" witch is uncommented.

Merci beaucoup.

radek8, 2022-10-15:
You have to connect the 10-pin cable to the EXP3 connector.

Send a photo of the wiring.
Send Marlin configuration files.

josgar84, 2022-10-15:
Hello,
as i mentioned above i have tried to connect the stock 10 pins cable to EXP3 and doesn't work at all.
here you are my config files. 

[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-TFT35-V3.0/files/9793198/Configuration.h.txt)

Thanks a lot

josgar84, 2022-10-16:
Hello some photos of the wiring
[image]
[image]
Have a nice day
José

radek8, 2022-10-16:
You misconnected the 10 pin connector.  On one side, turn it 180 degrees.

josgar84, 2022-10-16:
[image]

radek8, 2022-10-16:
There are several solutions to your problem.
- you can cut the protrusion as you write, or you can
- make a new cable with the correct orientation, or
- turn the opposite piece. (pull off the pins and put on in reverse)

[image]

josgar84, 2022-10-16:
Super i am going to try it
thanks

josgar84, 2022-10-16:
:(   I rotated  the connector (solution 3) as you said but the screen display "Marlin mode" and black screen .
More idea ? 
Merci

radek8, 2022-10-16:
Marlin's mod should work.
Send your display configuration
config.ini

josgar84, 2022-10-16:
I haven't got any .the firmware is stock firmware. I'm going to flash a new one and i'll go back tomorrow. I must buy a SD card.
Thank's and have a good evening (or a good day ?).
José

radek8, 2022-10-16:
In Marlin you have to enable
#define CR10_STOCKDISPLAY

You configure TFT with the Config.ini file.
you must have an active choice
marlin_type:0

josgar84, 2022-10-16:
IT works ! 
Your are brilliant ! 
Thanks a lot for your help.
José

josgar84, 2022-10-16:
Sorry i forgot for the others persons who have the same problem i set marlin_type: to 1 into Config.ini file of the screen.

closed 2022-10-16 by josgar84