bigtreetech/BIGTREETECH-TFT35-V3.0 #157: BMP size. 24bpp

DaLiV, 2022-10-24:
even 4bpp would be enough.
Ststistic:
* 24Bpp take 4.5Mb
* 8bpp take 1Mb and will be 3.5Mb free instead
* 4bpp - 700K - 3.8Mb free

smaller footprint of media gives more space for extended functionality possibility.

radek8, 2022-11-01:
Yes you are right.
The space occupied by the icons is not yet critical.
The display has two memory spaces.
There is approx. 230kB in the process, where the Program and eeprom data are stored. The second memory is an external SPI 8MB, where the font and icons are stored.
More critical is the space for the program and display functions. The space for icons is still sufficient. In an emergency, some icons can be recycled and use, for example, one icon for similar functions.
Icons with greater color depth look nicer.
It is possible that over time, if more functions are added and more icons are needed, the color depth of the icons will also be adjusted.

radek8, 2022-11-01:
Problems and improvements to the FW display are better addressed here:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues
The FW repository is tracked by multiple users and programmers.