bigtreetech/BIGTREETECH-TFT35-V3.0 #160: homing not working

Blisk, 2023-03-14:
I just cant make homing to work. I have installed BLtouch and I upload this firmware and still when I pres homing
it moves X and Y and Z goes up a bit and that is all.
I use BTT SKR mini 3.0 motherboard.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin

Any idea what to do to make it work?

Blisk, 2023-03-14:
I have updated again firmware with and now I get this error.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V3.0/Marlin/firmware-ender3-bltouch-for-z-homing.bin

stop called because of bltouch error - restart with m999

Blisk, 2023-03-14:
I manage to solve this with compiling marlin again and enabling
#define Z_SAFE_HOMING

radek8, 2023-04-01:
Close the ticket