bigtreetech/BIGTREETECH-TFT35-V3.0 #161: "No printer attached"

QUADMAN7, 2023-03-26:
I have a TFT35 connected to a Manta M5P running klipper firmware. I have connected it to the Manta m5p using the EXP1 and EXP2 connectorsseeing as the m5p doesn't have a TFT port. The problem i having is that it says there's no printer attached and I don't know why. Any help would be much appreciated. Thanks.

radek8, 2023-03-29:
The touch screen in touch mode communicates via the serial port on the TFT connector.
If you have not connected the TFT connector, the display cannot communicate and the printer will not connect.
Display FW is optimized for Marlin, Klipper not all functions will be available.
EXP1 and EXP2 connectors are for Marlin simulated mode communication

QUADMAN7, 2023-03-30:
I've tried using TFT cable and wiring it up to the main board that way but I still get a "no printer attached error". What should I do to get rid of this error?

radek8, 2023-03-30:
On the tisjarne you need to define the serial port and speed. The speed must be the same as on TFT.

QUADMAN7, 2023-03-30:
Will it not work with klipper and a Manta board even in the marlin simulated mode?

radek8, 2023-03-30:
I don't know. 
I have no experience with Klipper or Manta album. 
I know Marlin and TFT display.

QUADMAN7, 2023-03-30:
No solution was found