bigtreetech/BIGTREETECH-TFT35-V3.0 #162: I know this has probably been solved ... "outdated" icons upon firmware update!?

RobME3D, 2023-04-22:
I am a total newbie. I wont go into the details except that ...
I have repeatedly tried to update my BTT TFT 3.0 firmware.  I had a ton of problems and now, at least, I am as close as ... Outdated Icons.  What!?  I have spent HOURS on this ...
I have tried this a thousand times.
I have downloaded the latest BTT firmware.
I have reformatted the SD card a thousand times.  if is a 2GB card formated to FAT32.
I have tried different 2GB memory cards - no luck
formatted to FAT and FAT32 - not luck
The fonts load fine. The Config file loads fine. The language file loads fine ... just the Icons.
I have gone into the BMP directory and copied again - no luck
I went on Github and downloaded the latest and copied onto the memory card - no luck
I went in and compared the TFT35/bmp directory (144 files) to the larger screens (143 files) and taken out the logo_portrait.bmp - no luck.
I put it back in - no luck
The config file and the screen .bin file have uploaded (now *.cur) ....
I have no idea what to do now.  Help!

RobME3D, 2023-04-23:
Update:  I thought perhaps I had enabled a feature in the BTT SKR Mini E3 V3 firmware that needed an icon so I went in and added a custom_1.bmp (and 2 and 3) which were copies of the custom.bmp.  Still fails on the Icons: Update required.

radek8, 2023-04-25:
Can you post a photo of the error?

RobME3D, 2023-04-26:
> Can you post a photo of the error?
You bet!  Thanks ....
[image]
So I also went into the icon_list.inc file and individually verified that all of the icons called out in the X_ICON[name] command were in the bmp folder.  The only thing I think of is that I enabled/didn't enable something in the SKR Mini E3 V3 motherboard firmware that is causing a conflict?
For reference, here is what is on the TFT35 SD card:
[image]
[image]
[image]
I appreciate the assistance.

radek8, 2023-04-26:
Check the icon names. Set the file extension view on your computer and check if the files with the icon have the correct format:
name.bmp

RobME3D, 2023-04-27:
Thank You!  I thought I did that in creating the new SKR firmware but I will go back and make sure.
Thanks again for your help radek8 - super appreciated.