bigtreetech/BIGTREETECH-TFT35-V3.0 #164: TFT35 V3.0   self-switching that won't stop

Hor-tide, 2023-07-08:
Good day,
I would like to ask if my month old TFT v3.0 on the Ender 3 with the SKR Mini E3 v3.0 board has gone bad or if this error can be fixed.
When I turn on the printer I hear a beep and it keeps showing the menu and then the first line and then again. If I did it myself, it's fine.. but the display only started doing it after I ran the cables. I tried connecting the display directly to the board but it still happens. I disconnected the wires from RS232, it didn't help. But when I connected the original Endera 3 LCD instead of the TFT...the display works normally and I can move around it.
Can someone please advise me?

radek8, 2023-07-09:
What menus are switched?

Hor-tide, 2023-07-09:
The very first one. It all feels like when I turn on the printer and press and hold the button right after it starts. But, I tried the button and it is not pressed.

radek8, 2023-07-10:
You have a display embedded in a frame. The frame touches the touch surface of the display. Adjust the frame so that it does not clap and the display.

radek8, 2023-07-10:
Connect only the RS232 cable. 10 pin cable disconnect. Marlin mode won't work, but touch mode will be okay?

Hor-tide, 2023-07-10:
... 10 pin cable disconnect. Button light and little text "Marlin mode." And this is all. The display is only illuminated. Press button for maybe switch Touch mode - He doesn't respond.

Hor-tide, 2023-07-10:
I tried putting the reset.txt file on the card. After starting it asked me to calibrate the Touch screen. After 10 attempts, success. But the piezo still beeps but I can move around the screen. But I can't connect to the printer even after adjusting the speed. After connecting the 10 pin cable, the same thing.. "No printer attached."

Hor-tide, 2023-07-10:
Yes, i have already. After the first assembly of the board and the TFT display, everything worked... on the Touch, however, it printed backwards, because the values were meaningless, just like the course. But the Marlin worked great. Then I made an extension cord and it has been not-working ever since. Now I only use the original cables, but it didn't help.

radek8, 2023-07-11:
Did you install a new FW in the display? 
If so, what version did you use. 
What display top do you have? What is the designation of the processor on the display?

Hor-tide, 2023-07-11:
I tried using the reset.txt file procedure.
After that, I had to calibrate the screen several times before it was successful and I got into Touch mode. Since then I can move in it, but the beeping has not stopped. Can't even switch the mod to Marlin.
But a new problem appeared.. The TFT cannot connect to the printer.
Neither the new cable nor the correct speed helped.
I tried the new FW from GIThub - 3 times in total. I don't see the numbers, but I started chronologically from the original version.
The designation of the microprocessor is GD32F - the FW version was always GD-V3.x.x.x.
Thanks to the FW change, it is constantly showing me that the printer is not responding.
I tried to work with all cables: RS232 and 10pin, only RS232 or only 10pin. The only thing I found out is that after disconnecting the 10pin, the TFT stops beeping.

radek8, 2023-07-12:
The GD version is correct. 
Use the last version (master). 

Beeping generates Marlin on SKR..
Do you compile your own Marlin? 
If you hold your finger on the screen for 4 seconds, will the menu to switch modes appear?

radek8, 2023-07-12:
Send ini the file you used to configure the Display. 
Send the configuration files you used to compile Marlin.

Hor-tide, 2023-07-12:
If I hold my finger on the screen, the mode button works for me. (I thought I could reach the mode only with the button. Thanks for the information). When I switch to Marlin's mode, it starts switching offers again on its own with a constant beep. If I hold my finger on the screen again in Marlin, I can choose the mod again... so I jump to Touch mode.
Someone else is building the Marlin for me. But I have tested that it compiles very successfully / well.

radek8, 2023-07-12:
From your description of the glitch, it looks like the encoder is permanently pressed. 
It looks like a faulty BNT pin on the 10 pin connector, a faulty pin on the processor is a connected BNT pin, or a faulty encoder.

Hor-tide, 2023-07-12:
And can this error be fixed?

radek8, 2023-07-12:
[image]

radek8, 2023-07-12:
Zen BTN vs GND
And
BTN vs VCC 5V.

radek8, 2023-07-12:
If the display worked flawlessly and now has a problem with the serial port and encoder, it looks like the processor pins are damaged. 
If you were changing the wiring, there may be a problem with the cable, or you shorted something.

Hor-tide, 2023-07-12:
I will measure tomorrow.
Yes, it is likely that the error was caused by extending the cable or reversing it.

Hor-tide, 2023-07-13:
Today I bought a new TFT35 display.
I need to work with the printer and finish its assembly.
I am very glad that I found the energy here and the desire to help me solve my problem.
I appreciate that.
From what I read in other threads.. I should close this topic.

Thank you for your help.