bigtreetech/BIGTREETECH-TFT35-V3.0 #168: connect a TFT35 E3 to an Ender3 v4.2.7 mainboard

dbyy, 2024-01-07:
hello,
I installed the firmware on a TFT35 E3. 
I connected the 10-pin cable in 'exp3' on the display board and the other end in the TFT connector on the main-board.
After booting up the system, the display shows 'no printer attached'.
Do I have to connect also the 5-pin serial cable? And if so, where? There is no 5-pin connector on the main-board available?
Does that mean I cannot use the touch-screen feature on a v4.2.7 main-board?
Any help is much appreciated.

weaponrascal, 2024-01-16:
I have the same question and problem "How to connect to my 4.2.7 board" and get rid of that error message.

i did find on youtube a video that indicated steps to do just this.
In case you are interested, here is the link to it:  https://www.youtube.com/watch?v=kL9PJB7YFfA

Lets see if the process really works.

weaponrascal, 2024-01-16:
Well, I just reviewed the youtube video, not to much info regarding the No Printer Attached issue, did a good review of the board, the 3D printed enclosure and assembly.

Guess it is time to search some additional resources to try and et the answer.

dbyy, 2024-01-17:
Thanks for your reply. 
It is too bad there is no information about this problem. At least I haven't found one (yet). Keep looking.

closed 2024-01-17 by dbyy